kachurovskiy/lathecode #3: [Suggestion] Defining internal stock shape

Festivejelly, 2023-10-13:
It would be super handy to be able to specify the internal stock geometry.

At the moment if I want to do internal operations the tool assumes that I want to use the to cut all of it I think?

Take this model

; Machining Operations on 17mm Stock
; Using a 3mm wide parting off tool

UNITS MM
STOCK D17 
TOOL RECT R0.1 L1.5 ; Using a rectangular grooving tool

INSIDE
L1.75 D11
L1.5 D13.4
L6.75 D11

On paper this looks okay, the model looks like how I want it... but it actual fact it doesnt need to bore out the 11mm hole because thats already there. It just needs to cut the groove thats 1.5mm long. But the gcode that this generates does not look right.

kachurovskiy, 2023-10-14:
Yes, currently the editor only supports outside turning. Internal operations are not implemented because they're usually done with a tailstock that we don't control, require a lot of attention to the tool holder to avoid hitting the stock and generally used less. Hopefully one day they will be supported.

For the said groove, using H4 cut-off mode (negative pitch = internal) should do the job?

Do you have any other real-world examples of internal operations that would make sense to automate? Thanks!

Festivejelly, 2023-10-15:
It kinda does.

Here is what ive set up.

UNITS MM
STOCK D17 
TOOL RECT R0.1 L1.5 ; Using a rectangular grooving tool

INSIDE
L1.75 D11
L1.5 D13.4
L6.75 D11

This issue with this is that it doesnt really generate correct gcode.

kachurovskiy, 2023-10-15:
So what I have is:

- segment generation for `INSIDE` shapes - https://github.com/kachurovskiy/lathecode/blob/main/src/common/lathecode.ts#L98
- drawing of both inside and outside 3D shapes - https://github.com/kachurovskiy/lathecode/blob/main/src/scene.ts#L112
- drawing of both inside and outside 2D canvases - https://github.com/kachurovskiy/lathecode/blob/main/src/planner/painter.ts#L68

What's missing is:
- Ability to define stock with inside diameter
- Ability to define tool position / angle for inside operation (currently tool is always looking/cutting up)
- Planner logic for initial position of the tool for the inside operation - https://github.com/kachurovskiy/lathecode/blob/main/src/planner/plannerworker.ts#L48
- Planner logic for progressing through the stock for the inside operation - https://github.com/kachurovskiy/lathecode/blob/main/src/planner/plannerworker.ts#L126