kachurovskiy/lathecode #4: Ball end G-code generator - eliptic curve

KyrreKnive, 2023-11-14:
Hi Maxim,

I am trying to generate a ball end, but my input seems to generate an unevan curve on the left side.
STOCK D10
TOOL ROUND R0.2 L1.5

L4 DS0 DE8 CONV

L3.464 DS8 DE4 CONV ; 3.464 is the legth from center where the diameter = 4
L4 D4

kachurovskiy, 2023-11-14:
Did you mean RECT tool instead of ROUND?

First workaround is to define the full sphere but use the cutter that can't cut into the deep part e.g. parting blade:

[image]

Second workaround is draw the picture on figma.com with your exact dimensions (1px = 0.01mm) and use `From image` button.

KyrreKnive, 2023-11-15:
Thank's for the workarounds! 
Can you post the calculation (program lines) for the curve /elipse function (L3.464 DS8 DE4 CONV ) ?

kachurovskiy, 2023-11-15:
> Can you post the calculation (program lines) for the curve /elipse function (L3.464 DS8 DE4 CONV ) ?

@KyrreKnive is https://github.com/kachurovskiy/lathecode/blob/main/src/planner/painter.ts#L86 what you're looking for?

closed 2023-12-30 by kachurovskiy