kachurovskiy/lathecode #6: Tool shapes

Festivejelly, 2023-12-30:
Loving the tool so far it makes designing gcode an absolute breeze.

However my best tools are actually diamond shaped, I use ground sharp inserts designed for aluminium. But they work great on everything.

But with latheCode I can only define a rectangular or circular shaped tool.

I use DCMT and CCMT normally when facing:

https://www.shop-apt.co.uk/carbide-inserts-for-turning-uni-tip/dcgt-070202-alu-ak10-carbide-inserts-for-turning-ground-and-polished-for-aluminium-uni-tip.html
https://www.shop-apt.co.uk/carbide-inserts-for-turning-uni-tip/ccgt-060202-alu-ak10-carbide-inserts-for-turning-ground-and-polished-for-aluminium-uni-tip.html

with these tools

https://www.shop-apt.co.uk/apt-93-sdjcrl-lathe-turning-tools-for-dcmt-inserts/sdjcr-1010-f07-apt-lathe-turning-tool-for-dcmt-0702-inserts.html
[-](https://www.shop-apt.co.uk/apt-95-sclcrl-lathe-turning-tools-for-ccmt-inserts/sclcr-1010-h06-apt-lathe-turning-tool-for-ccmt-0602-inserts.html)

So ideally I want to be able to specify this diamond type tool so that it doesnt generate plunge cuts.

So essentially if diamond tool is selected do lots of surfacing and facing to get to dimension

kachurovskiy, 2023-12-30:
I've added support for angled tools, please give it a try.

Festivejelly, 2023-12-30:
Sweet lord that was quick ill give it a test :)

Festivejelly, 2023-12-31:
That doesnt seem to be working for me. Ive specified the tool as per the readme but it still tries to make the plunge cuts:

UNITS MM
STOCK D12

TOOL ANG R0.2 L7.75 A30 NA55

L5.5 DS0 DE4
L1.5 D8
L1 D10.5

kachurovskiy, 2023-12-31:
Cut logic is not dependent on the tool at the moment. It's always doing face cutting to reduce part deflection. Cutting e.g. a chess figure would be much less feasible if we thin out the part first. I'm open to supporting multiple cut strategies in the future but I wonder about real-world benefits of turning vs. plunging here.

Festivejelly, 2023-12-31:
Ah I see how you mean now. I was thinking it was going to try plunging in the middle of the stock but it wont. I think plunge cuts with a surfacing pass will absolutely be sufficient here.

Festivejelly, 2024-01-04:
Looking at the finish that plunge cuts do it really isnt as good as turning passes.

Take a cone plunging in multiple steps then doing a very thin finishing pass yields ugly results.

[image]

I hand coded this Gcode and it yields a much better surface finish. Maybe we could specify the depth of a finishing pass?

[image]

kachurovskiy, 2024-01-04:
Can you please share the lathecode for this part?

Festivejelly, 2024-01-04:
Sure its:

UNITS MM
STOCK D12

TOOL RECT R0.2 L3

L5.5 DS0 DE4
L1.5 D8
L1 D10.5

kachurovskiy, 2024-01-05:
Here's the toolpath I have generated for it with my local changes for the finish pass:

[image]

Finish depth by default is 0.1mm but can be adjusted. Would that work?

kachurovskiy, 2024-01-07:
@Festivejelly finish pass should work now, please give it a try.

Festivejelly, 2024-01-07:
@kachurovskiy this works really well thanks for adding this. I like the animation too, nice touch! Given you've added support for the other tools and depth of cut control I think this one can be closed now. Nice work.

closed 2024-01-07 by kachurovskiy