kachurovskiy/lathecode #9: [Bug] Local Storage issue

Festivejelly, 2024-01-21:
At the moment when you click more under the lathe code to be able to save and load to local storage it will show anything in local storage in the list. This could include local storage from plugins and extensions etc.

To fix this I think maybe the names in local storage should be prefixed with latheCode_{name of save}

And when loading the page iterate only over those that has latheCode_ prefix.

kachurovskiy, 2024-01-21:
Well, some folks already have stuff saved without the prefix. We can filter and only show multi-line values.

Festivejelly, 2024-01-21:
Could we maybe iterate over existing values to determine if they are lathe code, then “fix” the ones that are

kachurovskiy, 2024-01-25:
Yeah, but then we'd have to keep that migration code around for some unknown period of time. Hope the issue is sufficiently mitigated :) Thanks!

closed 2024-01-25 by kachurovskiy