kachurovskiy/lathecode #12: [Bug] Chamfers in one direction not completing properly

Festivejelly, 2024-03-11:
Im getting an odd issue where chamfers moving from right to left dont seem to be getting cut, but chamfers moving left to right look fine

Here is a model to reproduce:

UNITS MM
STOCK D16.9 
TOOL RECT R0.2 L2 ; Using a rectangular grooving tool

L0.5 DS14.5 DE15.5
L12 D15.5
L0.5 DS15.5 DE14.5
L5 D10
L0.5 DS14.5 DE15.5
L8.5 D15.5
L0.5 DS15.5 DE14.5
L3 D10

Is it a case of the tool dimension might not be 100% accurate?
[image]

kachurovskiy, 2024-03-11:
Not sure I see the problem... Toolpath seems to match the model well?

[image]

[image]

Festivejelly, 2024-03-11:
Thats what confused me too. Preview shows fine. I thought maybe it was a backlash issue? But then all the dimensions seem to measure accurately so im a bit confused as to how it would be off. Maybe backlash isnt working well for small moves.

kachurovskiy, 2024-03-11:
Can you post the picture of the resulting part or maybe even how the finish pass went?

Festivejelly, 2024-03-11:
Will do, i'll take some pics when i go into my garage later.

Festivejelly, 2024-03-11:
surface quality was really good! the finishing pass worked super well on the other chamfers just not the ones I highlighted. It kinda misses them. Maybe I need a deeper finishing pass.

kachurovskiy, 2024-03-11:
Yes, likely a backlash issue. Consider remeasuring the backlash? Felix reported similar problem. Alternatively, adjust the lathecode to make a deeper chamfer in the areas where it's insufficient.

Festivejelly, 2024-03-13:
Closing this as Im 100% sure this was due to backlash. Ive now properly measured what my backlash is and added that to my controller.

closed 2024-03-13 by Festivejelly