kachurovskiy/lathecode #15: [Bug] Finishing pass not being applied.

Festivejelly, 2024-03-17:
Take this:

UNITS MM
STOCK D16
TOOL RECT R0.2 L2 ; Using a rectangular grooving tool
DEPTH CUT0.5 FINISH0.2

L0.5 D0
L5 D10
L5 D15

When I look at the generated gcode using my visualiser I can see that it only cuts as deep as 5.2mm ABS. Which means its not completing the finishing passes.

https://festivejelly.github.io/LatheGcodeVisualiser/

[image]

I can see on the planner it looks like it knew it should do it:

[image]
 
 
 But it didnt generate the toolpath for it.

kachurovskiy, 2024-03-17:
@Festivejelly I suggest to stop using `D0` suffix in your lines :) If you specify `L0.5` instead of `L0.5 D0` it works as intended.

Festivejelly, 2024-03-17:
@kachurovskiy Ahhh I was wondering why sometimes it worked and sometimes not, thats probably why,

Festivejelly, 2024-03-17:
Can confirm this now works as of your commit so ill close. Cheers.

closed 2024-03-17 by Festivejelly