kachurovskiy/lathecode #16: Pecking when parting

Festivejelly, 2024-06-01:
I was making this today:

UNITS MM
STOCK D12

TOOL RECT R0.2 L2
DEPTH CUT0.5 FINISH0.2

L0.5
L5.5 DS0 DE4
L0.2 DS7.8 DE8
L1.3 D8
L0.2 DS10.3 DE10.5
L0.7 D10.5
L0.2 DS10.5 DE10.3
L2

Its a stainless steel cone. And it was all going great until I went to part it. Then my whole lathe shook to pieces! I had to slow the RPM right down to like 250 and manually peck away.

Is there a way we can define a pecking routine for parting or deep grooves?

Also notice that the fillet at the back edge doesnt get cut, but if you do:

UNITS MM
STOCK D12

TOOL RECT R0.2 L2
DEPTH CUT0.5 FINISH0.2

L0.5
L5.5 DS0 DE4
L0.2 DS7.8 DE8
L1.3 D8
L0.2 DS10.3 DE10.5
L0.7 D10.5
L0.2 DS10.5 DE10.3
L2 D5

Then it does cut it.

kachurovskiy, 2024-06-02:
Hi @Festivejelly, nice model. Yes, looks like there was a problem when the last few mm width of the finish layer wouldn't be cut if it was on the left edge of the part - requiring the tool to travel into the to-be-parted area of the part.

[image]

That piece of material should now be removed during parting.

Festivejelly, 2024-06-02:
Might be of interest to you as I saw you had NC viewer open which I couldnt get on with so i hacked my own together:

https://festivejelly.github.io/LatheGcodeVisualiser/

If you paste the below in the simulate section and click simulate you can use the slider to step over the cuts and travels etc.

; UNITS MM
; STOCK D12
; 
; TOOL RECT R0.2 L2
; DEPTH CUT0.5 FINISH0.2
; 
; L0.5
; L5.5 DS0 DE4
; L0.1 DS7.8 DE8
; L1.4 D8
; L0.1 DS10.3 DE10.5
; L0.8 D10.5
; L0.1 DS10.5 DE10.3
; L2 D9

; Run time 2.5 min, cutting 77% of time
; Working area 10.50 by 6.14 mm

G21 ; metric
G18 ; ZX plane
G90 ; absolute positioning
F200
X-6 ; zero your tool X on centerline
Z0 ; zero your tool Z at the right edge of the stock
G91 ; relative positioning
Z0.5

F50
X6.14 ; cut 3 mm2

F200
X-6.14
Z0.5

F50
X5.87 ; cut 2.9232 mm2
Z-0.31 X0.12 ; cut 0.0217 mm2 at -68.84°

F200
X-5.99
Z0.81

F50
X5.69 ; cut 2.8329 mm2
Z-0.43 X0.16 ; cut 0.024 mm2 at -69.59°

F200
X-5.85
Z0.93

F50
X5.51 ; cut 2.7429 mm2
Z-0.4 X0.15 ; cut 0.0231 mm2 at -69.44°

F200
X-5.66
Z0.9

F50
X5.33 ; cut 2.6526 mm2
Z-0.44 X0.16 ; cut 0.0225 mm2 at -70.02°

F200
X-5.49
Z0.94

F50
X5.15 ; cut 2.5626 mm2
Z-0.41 X0.15 ; cut 0.0217 mm2 at -69.90°

F200
X-5.3
Z0.91

F50
X4.97 ; cut 2.4723 mm2
Z-0.39 X0.14 ; cut 0.021 mm2 at -70.25°

F200
X-5.11
Z0.89

F50
X4.78 ; cut 2.3781 mm2
Z-0.4 X0.15 ; cut 0.0236 mm2 at -69.44°

F200
X-4.93
Z0.9

F50
X4.6 ; cut 2.2879 mm2
Z-0.4 X0.15 ; cut 0.0236 mm2 at -69.44°

F200
X-4.75
Z0.9

F50
X4.42 ; cut 2.1978 mm2
Z-0.41 X0.15 ; cut 0.0227 mm2 at -69.90°

F200
X-4.57
Z0.91

F50
X4.24 ; cut 2.1076 mm2
Z-0.41 X0.15 ; cut 0.0221 mm2 at -69.90°

F200
X-4.39
Z0.91

F50
X4.06 ; cut 2.0175 mm2
Z-0.42 X0.15 ; cut 0.0213 mm2 at -70.35°

F200
X-4.21
Z0.92

F50
X2 ; cut 0.9917 mm2

F200
Z-0.28 X0.01 ; cut 0.0009 mm2 at -87.95°

F50
Z-0.06 X0.04 ; cut 0.0024 mm2 at -56.31°
Z-0.11 X0.11 ; cut 0.0012 mm2 at -45.00°

F200
X-2.16
Z0.95

F50
X2 ; cut 1 mm2

F200
X-2
Z0.5

F50
X2 ; cut 1 mm2

F200
X-2
Z0.5

F50
X0.75 ; cut 0.3667 mm2

F200
Z-0.28 X0.01 ; cut 0.0009 mm2 at -87.95°

F50
Z-0.06 X0.04 ; cut 0.0024 mm2 at -56.31°
Z-0.11 X0.11 ; cut 0.0012 mm2 at -45.00°

F200
X-0.91
Z0.95

F50
X0.75 ; cut 0.3741 mm2

F200
X-0.75
Z0.5

F50
X0.75 ; cut 0.3749 mm2

F200
X-0.75
Z0.5

F50
X0.75 ; cut 0.375 mm2

F200
X-0.75
Z0.5

F50
X0.75 ; cut 0.375 mm2

F200
X-0.75
Z0.5

F10
X1.3 ; cut 1.4677 mm2

F200
X-1.3
Z-10.49
X6.14

F50
Z0.52 ; cut 0.0052 mm2

F200
Z0.02 X-0.05 ; cut 0.0005 mm2 at -21.80°
Z0.05 X-0.05 ; cut 0.0006 mm2 at -45.00°
Z0.09 X-0.05 ; cut 0.0009 mm2 at -60.95°

F50
Z5.31 X-1.93 ; cut 0.0531 mm2 at -70.03°
X-1.81 ; cut 0.0181 mm2

F200
Z0.02 X-0.05 ; cut 0.0005 mm2 at -21.80°

F50
Z0.15 X-0.15 ; cut 0.0016 mm2 at -45.00°

F200
Z0.08 X-0.05 ; cut 0.0008 mm2 at -57.99°

F50
Z1.25 ; cut 0.0125 mm2
X-1 ; cut 0.01 mm2

F200
Z0.02 X-0.05 ; cut 0.0005 mm2 at -21.80°

F50
Z0.15 X-0.15 ; cut 0.0016 mm2 at -45.00°

F200
Z0.08 X-0.05 ; cut 0.0008 mm2 at -57.99°

F50
Z2.5 ; cut 0.025 mm2
Z0.05 X0.02 ; cut 0.0019 mm2 at 68.20°
Z0.15 X0.15 ; cut 0.0036 mm2 at 45.00°

F200
Z0.05 X0.08 ; cut 0.0008 mm2 at 32.01°

;peck parting

F10
X1 ; cut

F100
X-1 ; retract
X1 ; travel

F10
X1 ; cut
F100
X-1 ; retract
X1 ; travel

F10
X1 ; cut
F100
X-1 ; retract
X1 ; travel

F10
X1 ; cut
F100
X-1 ; retract
X1 ; travel

F10
X1 ; cut
F200
X-6; retract

F200
Z-10.5

kachurovskiy, 2024-06-03:
Very nice GCode visualization in your tool! Love that you show the next move with the dashed line. It's a little hard to see black on grey text though 😄

[image]

Did you try different speeds for parting, perhaps a combination of very fast spindle and medium speed cutter advance could get you a good result in 1 parting pass?

Festivejelly, 2024-06-03:
Oops CSS bug, I think its to do with system colours etc ill need to look at that!

I think a big part of the problem is that I was machining 304. I have 303 so im going to try that.

It was cutting everything else fine, maybe ill swap to my 1.5mm parting tool.

closed 2025-02-09 by kachurovskiy