kachurovskiy/lathecode #18: [Suggestion] Prefer profiling cuts

Festivejelly, 2024-06-07:
I wonder if there is a way we could specify a preferncing to doing profiling cuts for long areas.

For example take this:

UNITS MM
STOCK D15

TOOL RECT R0.2 L2

L0.5
L3 D1.93
L3 DS2.5 DE6 CONC
L1.73 D12.55
L2.2 D9.5
L33.5 D12.55
L2 D0

It results in a lot of cuts into the work, and a lot of retractions.

For bringing down the out diameter I think doing a long profile cut would be quick as it results in fewer retractions.

The material im using a brass so it would be quite for giving to profiling cuts that are maybe a bit deeper than normal.

I can kinda simulate what I want by making a the finishing pass very deep:

UNITS MM
STOCK D15

TOOL RECT R0.2 L2
DEPTH CUT1 FINISH2

L0.5
L3 D1.93
L3 DS2.5 DE6 CONC
L1.73 D12.55
L2.2 D9.5
L33.5 D12.55
L2 D0

But this problem here is I dont think my tools could take that. So ideally this would be done over several passes

kachurovskiy, 2024-06-07:
Yeah, that's a good idea, make the cut logic adjustable with a command - initially I actually had the passes done length-wise but switched to facing cuts as they cover more tool and part combinations.

It would be even better to do zig-zag passes which would allow to almost completely eliminate non-cutting movements.

Festivejelly, 2024-06-10:
I think rather than trying to detect which strategy to use, maybe it could be a command as you suggested? Could be tricky to determine if its better to do profiling passing or plunging cuts.
Would give users a bit more control too over how they want to do it.

I havent heard of zig zag passes ill have to look into that.

Festivejelly, 2025-02-09:
@kachurovskiy Glad to see this being added. When I try to specify a depth of cut though I get an error.