kachurovskiy/lathecode #19: tool selection

kotovasia5120, 2024-11-08:
I read the manual but I don't understand what the editor wants from me.
```
TOOL ANG R0.4 L11.6 A32.5 NA55
DEPTH CUT1 FINISH0.2
FEED MOVE200 PASS50 PART10

STOCK D20
L3 R6
L0.5 RS6 RE5.51 CONV
L2.5 RS5.5 RE3 CONC
L1 R3
L2.5 RS3 RE5.49 CONC
L0.5 RS5.5 RE6 CONV
L3.3 RS6 RE0 CONV
L4 R1
```
[image]

kotovasia5120, 2024-11-08:
Also if you specify instead of L2.5 RS3 RE5.49 CONC, L2.5 RS3 RE5.5 CONC then the model is incorrect

kachurovskiy, 2024-11-09:
`STOCK` needs to go before `TOOL`.

What do you mean by broken? Seems fine. Please post screenshots.

[image]

kotovasia5120, 2024-11-09:
> `STOCK` needs to go before `TOOL`.

I didn't see this in the manual. Maybe add it or make the order tolerable?
> What do you mean by broken? Seems fine. Please post screenshots.

The rounding is lost
[image]
[image]

kachurovskiy, 2024-11-09:
Yeah, looks like a bug somewhere in the optimization logic.

kotovasia5120, 2024-11-09:
> Yeah, looks like a bug somewhere in the optimization logic.

By the way, freecad also often has problems with such connections. This problem can be serious.

kotovasia5120, 2024-11-11:
Maybe I'm wrong, but for this part, the code is output on the d16 rod instead of d32. But maybe I didn't figure out the bindings.

closed 2025-02-09 by kachurovskiy