kachurovskiy/lathecode #20: h4 stop at the same place in the program.

kotovasia5120, 2024-12-26:
This part was fine before. Now it stops.
I tried changing the speed and the amount of feed per pass, but nothing changed.
```
STOCK D14.5

;sdnc1212h0 dc07 /\
;TOOL ANG R0.4 L7.8 A0 NA55
;mgehr1212-2 mgmn200 ||
TOOL RECT R0.2 L2
;Metal
DEPTH CUT0.9 FINISH0.2
FEED MOVE150 PASS5 PART5
;PA
;DEPTH CUT1 FINISH0.2
;FEED MOVE300 PASS100 PART10

L2.5 R6
L0.5 RS6 RE5.51 CONV
L2.5 RS5.5 RE3 CONC
L1 R3
L2.5 RS3 RE5.49 CONC
L0.5 RS5.5 RE6 CONV
L3.3 RS6 RE0 CONV
L2.5 R2
```
```
; STOCK D14.5
; 
;sdnc1212h0 dc07 /\
;TOOL ANG R0.4 L7.8 A0 NA55
;mgehr1212-2 mgmn200 ||
; TOOL RECT R0.2 L2
;Metal
; DEPTH CUT0.9 FINISH0.2
; FEED MOVE150 PASS5 PART5
;PA
;DEPTH CUT1 FINISH0.2
;FEED MOVE300 PASS100 PART10
; 
; L2.5 R6
; L0.5 RS6 RE5.51 CONV
; L2.5 RS5.5 RE3 CONC
; L1 R3
; L2.5 RS3 RE5.49 CONC
; L0.5 RS5.5 RE6 CONV
; L3.3 RS6 RE0 CONV
; L2.5 R2

; Run time 16.3 min, cutting 96% of time
; Working area 15.30 by 5.25 mm

G21 ; metric
G18 ; ZX plane
G90 ; absolute positioning
F150
X-7.25 ; zero your tool X on centerline
Z0 ; zero your tool Z at the right edge of the stock
G91 ; relative positioning
Z0.9

F5
X1.05 ; cut 0.9377 mm2

F150
X-1.05
Z0.9

F5
X1.05 ; cut 0.945 mm2

F150
X-1.05
Z0.9

F5
X1.05 ; cut 0.945 mm2

F150
X-1.05
Z0.9

F5
X1.05 ; cut 0.945 mm2

F150
X-1.05
Z0.9

F5
X1.08 ; cut 1.0008 mm2

F150
X-1.08
Z0.9

F5
X3.04 ; cut 4.9033 mm2
Z-0.07 X-0.11 ; cut 0.0134 mm2 at 32.47°
Z-0.18 X-0.37 ; cut 0.0902 mm2 at 25.94°
Z-0.1 X-0.33 ; cut 0.1262 mm2 at 16.86°
Z-0.03 X-0.22 ; cut 0.0913 mm2 at 7.77°
Z-0.02 X-0.04 ; cut 0.0173 mm2 at 26.57°
Z-0.01 X-0.38 ; cut 0.1588 mm2 at 1.51°
Z-0.1 X-0.23 ; cut 0.0678 mm2 at 23.50°
Z-0.13 X-0.15 ; cut 0.0132 mm2 at 40.91°
Z-0.12 X-0.08 ; cut 0.0053 mm2 at 56.31°
Z-0.1 X-0.04 ; cut 0.0021 mm2 at 68.20°

F150
X-1.09
Z1.76

F5
X3.82 ; cut 4.277 mm2
Z-0.25 X-0.14 ; cut 0.0766 mm2 at 60.75°
Z-0.23 X-0.17 ; cut 0.1118 mm2 at 53.53°
Z-0.29 X-0.29 ; cut 0.1052 mm2 at 45.00°
Z-0.13 X-0.17 ; cut 0.0183 mm2 at 37.41°

F150
X-3.05
Z1.8

F5
X3.99 ; cut 3.7705 mm2
Z-0.18 X0.04 ; cut 0.1035 mm2 at -77.47°
Z-0.21 X-0.03 ; cut 0.0342 mm2 at 81.87°
Z-0.33 X-0.1 ; cut 0.0272 mm2 at 73.14°
Z-0.09 X-0.04 ; cut 0.0012 mm2 at 66.04°

F150
X-3.86
Z1.71

F5
X3.57 ; cut 3.2053 mm2
Z-0.24 X0.16 ; cut 0.1031 mm2 at -56.31°
Z-0.38 X0.18 ; cut 0.0597 mm2 at -64.65°
Z-0.17 X0.06 ; cut 0.0043 mm2 at -70.56°

F150
X-3.97
Z1.69

F5
X1.98 ; cut 1.7709 mm2
Z-0.14 X0.57 ; cut 0.4494 mm2 at -13.80°
Z-0.22 X0.44 ; cut 0.253 mm2 at -26.57°
Z-0.15 X0.21 ; cut 0.0755 mm2 at -35.54°
Z-0.34 X0.34 ; cut 0.0385 mm2 at -45.00°

F150
X-3.54
Z1.75

F5
X1.05 ; cut 0.9377 mm2
Z-0.34 X0.01 ; cut 0.004 mm2 at -88.32°
Z-0.21 X0.08 ; cut 0.022 mm2 at -69.15°
Z-0.06 X0.04 ; cut 0.0071 mm2 at -56.31°
Z-0.14 X0.14 ; cut 0.0127 mm2 at -45.00°
Z-0.1 X0.17 ; cut 0.0035 mm2 at -30.47°

F150
Z-0.02 X0.06 ; cut 0.0003 mm2 at -18.43°
X-1.55
Z1.77

F5
X1.05 ; cut 0.945 mm2

F150
X-1.05
Z0.9

F5
X1.09 ; cut 1.0203 mm2
Z-0.17 X-0.02 ; cut 0.0015 mm2 at 83.29°

F150
X-1.07
Z1.07

F5
X1.49 ; cut 1.7727 mm2
Z-0.12 X-0.08 ; cut 0.0269 mm2 at 56.31°
Z-0.33 X-0.18 ; cut 0.0456 mm2 at 61.39°
Z-0.17 X-0.07 ; cut 0.0093 mm2 at 67.62°
Z-0.21 X-0.06 ; cut 0.0051 mm2 at 74.05°

F150
X-1.1
Z1.73

F5
X2.41 ; cut 3.1925 mm2
Z-0.26 X-0.34 ; cut 0.0834 mm2 at 37.41°
Z-0.24 X-0.26 ; cut 0.1222 mm2 at 42.71°
Z-0.39 X-0.32 ; cut 0.0561 mm2 at 50.63°

F150
X-1.49
Z1.79

F5
X4.34 ; cut 6.0499 mm2
Z-0.25 X-0.7 ; cut 0.1184 mm2 at 19.65°
Z-0.25 X-0.57 ; cut 0.2664 mm2 at 23.68°
Z-0.24 X-0.41 ; cut 0.2036 mm2 at 30.34°
Z-0.16 X-0.24 ; cut 0.0169 mm2 at 33.69°

F150
X-2.42
Z1.8

F5
X5.05 ; cut 5.2967 mm2
Z-0.72 ; cut 0.2801 mm2
Z-0.17 X-0.68 ; cut 0.2505 mm2 at 14.04°

F150
X-4.37
Z-14.41
X1.4

F5
Z0.02 X-0.05 ; cut 0.0031 mm2 at -21.80°
Z0.05 X-0.05 ; cut 0.0081 mm2 at -45.00°
Z0.08 X-0.05 ; cut 0.0115 mm2 at -57.99°
Z4.21 ; cut 0.5443 mm2
Z0.13 X0.02 ; cut 0.0067 mm2 at 81.25°
Z0.2 X0.09 ; cut 0.0175 mm2 at 65.77°
Z0.13 X0.11 ; cut 0.0105 mm2 at 49.76°
Z0.07 X0.09 ; cut 0.0034 mm2 at 37.87°
Z0.06 X0.12 ; cut 0.0021 mm2 at 26.57°
Z0.05 X0.21 ; cut 0.0021 mm2 at 13.39°
Z0.01 X0.31 ; cut 0.0057 mm2 at 1.85°
Z0.07 X0.36 ; cut 0.0178 mm2 at 11.00°
Z0.09 X0.25 ; cut 0.0201 mm2 at 19.80°
Z0.12 X0.26 ; cut 0.0289 mm2 at 24.78°
Z0.32 X0.44 ; cut 0.0711 mm2 at 36.03°
Z0.32 X0.28 ; cut 0.071 mm2 at 48.81°
Z0.13 X0.09 ; cut 0.0311 mm2 at 55.30°
Z0.33 X0.18 ; cut 0.1225 mm2 at 61.39°
Z0.22 X0.08 ; cut 0.1549 mm2 at 70.02°
Z0.41 X0.09 ; cut 0.2231 mm2 at 77.62°
Z0.36 X-0.08 ; cut 0.0712 mm2 at -77.47°
Z0.23 X-0.09 ; cut 0.0483 mm2 at -68.63°
Z0.42 X-0.23 ; cut 0.0781 mm2 at -61.29°
Z0.19 X-0.15 ; cut 0.0347 mm2 at -51.71°
Z0.34 X-0.38 ; cut 0.0615 mm2 at -41.82°
Z0.18 X-0.28 ; cut 0.0319 mm2 at -32.74°
Z0.12 X-0.28 ; cut 0.0211 mm2 at -23.20°
Z0.07 X-0.24 ; cut 0.0121 mm2 at -16.26°
Z0.06 X-0.29 ; cut 0.0082 mm2 at -11.69°
Z0.01 X-0.31 ; cut 0.0033 mm2 at -1.85°
Z0.02 X-0.14 ; cut 0.0022 mm2 at -8.13°
Z0.08 X-0.18 ; cut 0.0131 mm2 at -23.96°
Z0.05 X-0.07 ; cut 0.0087 mm2 at -35.54°
Z0.21 X-0.17 ; cut 0.0367 mm2 at -51.01°
Z0.14 X-0.06 ; cut 0.0246 mm2 at -66.80°
Z0.14 X-0.03 ; cut 0.0239 mm2 at -77.91°
Z1.72 ; cut 0.0965 mm2
Z0.01 X0.01 ; cut 0.0067 mm2
Z0.23 X0.01 ; cut 0.0083 mm2 at 87.51°
Z0.41 X0.1 ; cut 0.0486 mm2 at 76.29°
Z0.5 X0.25 ; cut 0.0762 mm2 at 63.43°
Z0.32 X0.24 ; cut 0.0518 mm2 at 53.13°
Z0.16 X0.14 ; cut 0.0249 mm2 at 48.81°
Z0.3 X0.34 ; cut 0.0498 mm2 at 41.42°
Z0.24 X0.32 ; cut 0.0384 mm2 at 36.87°
Z0.15 X0.23 ; cut 0.0235 mm2 at 33.11°
Z0.27 X0.49 ; cut 0.0418 mm2 at 28.86°
Z0.25 X0.58 ; cut 0.0374 mm2 at 23.32°
Z0.24 X0.72 ; cut 0.0336 mm2 at 18.43°
Z0.14 X0.57 ; cut 0.4009 mm2 at 13.80°
Z0.72 ; cut 0.1503 mm2
X-5.25 ; cut 0.0537 mm2

F150
Z-15.3
```

kotovasia5120, 2024-12-26:
[image]

kotovasia5120, 2024-12-26:
I also changed several esp and firmware versions.

kachurovskiy, 2024-12-26:
Try doubling `GCODE_WAIT_EPSILON_STEPS`? Most likely we're getting stuck in https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L3109 on that `Z-0.01`

https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L89

kotovasia5120, 2024-12-26:
> Try doubling `GCODE_WAIT_EPSILON_STEPS`

put 20. the stop occurred in the same place

kachurovskiy, 2024-12-26:
I have a suspicion that one of the speeds in https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L3065-L3067 is getting set to 0. Try adding at the end of that function:

```
  if (x.speedMax <= 0) x.speedMax = 1;
  if (z.speedMax <= 0) z.speedMax = 1;
  if (a1.speedMax <= 0) a1.speedMax = 1;
```

kotovasia5120, 2024-12-26:
> У меня есть подозрение, что одна из скоростей в https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L3065-L3067 устанавливается на 0. Попробуйте добавить в конец этой функции:
[image]

Thank you. It worked. I'm surprised how quickly you found the problematic place from the description.
Now between some lines of gcode there was about 20 seconds of downtime.

Is it because of the low feed rate?

kachurovskiy, 2024-12-27:
Yeah, instead of `1` we should use `20` or something like...

```
  if (x.speedMax < minStepsPerSecX) x.speedMax = minStepsPerSecX;
  if (z.speedMax < minStepsPerSecZ) z.speedMax = minStepsPerSecZ;
  if (a1.speedMax < minStepsPerSecA1) a1.speedMax = minStepsPerSecA1;
```

closed 2024-12-28 by kotovasia5120

kachurovskiy, 2025-01-13:
@kotovasia5120 did you ever try the latest suggestion? If you did, please let me know if it works for you.

kotovasia5120, 2025-01-14:
It worked.

> @kotovasia5120 did you ever try the latest suggestion? If you did, please let me know if it works for you.