kachurovskiy/lathecode #22: [Suggestion] Continue machining with different tool

Festivejelly, 2025-03-13:
It would be handy to have the lathecode generator be able to generate code for multiple tools.

Lets say for example you wanted to machine this:

UNITS MM
STOCK D23.25
TOOL ANG R0.4 L11.6 A32.5 NA55
DEPTH CUT0.5 FINISH0.2
FEED MOVE200 PASS50 PART10

L2 DS9.4 DE11
L4 DS11 DE12.5
L7 D12.5
L13 D9.8
L2

This sort of tool leaves a good finish but it cannot cut the straight groove at the back and it cant part off.

So we need to swap to a different tool such as the rectangular parting tool:

UNITS MM
STOCK D23.25
TOOL RECT R0.2 L2
DEPTH CUT0.5 FINISH0.2
FEED MOVE200 PASS50 PART10

L2 DS9.4 DE11
L4 DS11 DE12.5
L7 D12.5
L13 D9.8
L2

But of course we dont want it to cut all of the profile again as it would waste time.

So why not a hybrid?

UNITS MM
STOCK D23.25
STARTING TOOL ANG R0.4 L11.6 A32.5 NA55
FINISHING TOOL RECT R0.2 L2
DEPTH CUT0.5 FINISH0.2
FEED MOVE200 PASS50 PART10

L2 DS9.4 DE11
L4 DS11 DE12.5
L7 D12.5
L13 D9.8
L2

This way it will generate maybe 2 sets of gcode? The first set cuts as normal like above using the starting tool. And a 2nd set of gcode for the finishing tool.

So you would run that Gcode first. Then swap over the tool and run the 2nd set of gcode.

kachurovskiy, 2025-03-14:
The simple solution here is to have 2 lathecodes, leave a piece of material to zero off after the first tool is done.

You could also try another tool e.g. MGGN200, I heard it leaves a good finish, can get into your corner and doesn't leave a puck when parting.

Festivejelly, 2025-03-14:
I do usually use MGGN200 for things like brass and Aluminium but for stainless steel it doesnt work so great, its not bad but its not as good as other inserts.

I think I get where you're coming from with the 2 lathecodes though. For the second lathecode I could "fake" whats left so it cuts in the right areas.